doe300/VC4C #166: VC4C build error (make compile error)

jjsshh3116, 2024-01-04:
Hi! I try build openCL on raspberry pi3 b+ board. But occur error when during make at VC4C/build  directory. I follow this post: (https://qengineering.eu/install-opencl-on-raspberry-pi-3.html) 

Error message: 
[image]

And ㅅhis is the execution environment version.
[image]

Thank you for reading my article.

kotocom, 2024-01-17:
> Hi! I try build openCL on raspberry pi3 b+ board. But occur error when during make at VC4C/build directory. I follow this post: (https://qengineering.eu/install-opencl-on-raspberry-pi-3.html)
> 
> Error message: <img alt="Screenshot 2024-01-04 at 10 47 29" width="1500" src="https://private-user-images.githubusercontent.com/49449767/294071521-7ac49056-aebd-4545-bcc4-6808535636a7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU1MjYzMjYsIm5iZiI6MTcwNTUyNjAyNiwicGF0aCI6Ii80OTQ0OTc2Ny8yOTQwNzE1MjEtN2FjNDkwNTYtYWViZC00NTQ1LWJjYzQtNjgwODUzNTYzNmE3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE3VDIxMTM0NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI1ZTI4Nzk5YTVhNDkyMjJhZjM5Yzc1Yjc4MzYzMWMwYTI4YzEzYmNmZDAwMWFmNDhiNGQzZjhmMmNiM2FkMDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.uXJoCQQZ-mSH3cHzzYbew_bIvZ4CzTHJmJX_mk246eg">
> 
> And this is the execution environment version. <img alt="Screenshot 2024-01-04 at 11 25 08" width="1001" src="https://private-user-images.githubusercontent.com/49449767/294071678-bdc8d438-aca4-4be7-b105-c2f28649fcc4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU1MjYzMjYsIm5iZiI6MTcwNTUyNjAyNiwicGF0aCI6Ii80OTQ0OTc2Ny8yOTQwNzE2NzgtYmRjOGQ0MzgtYWNhNC00YmU3LWIxMDUtYzJmMjg2NDlmY2M0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE3VDIxMTM0NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTYxNzBkZTNkODc2ZWMzNTg3NDZmZmNkNmJjNzc5NDg0MzRiNGQ3ZjEwODIyYTg0YmUzNmRmNWM1OTJlZjg2NWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.HCTjBYFj3b4NLZtTYI3FNq2peYrc74BYcC9GNUWBXSc">
> 
> Thank you for reading my article.

Yeah. Same problem.

I'd recommend you just comment out the lines 827,837, 980 in the ../VC4C/test/TestInstructions.cpp and ignore/recompile

aruzhanabdirashova, 2024-01-21:
I had the same problem, and commented out. But now clinfo shows no available platforms

jjsshh3116, 2024-01-26:
> I had the same problem, and commented out. But now clinfo shows no available platforms

Right, When I do the same clinfo, 0 comes out.

Bravo13, 2024-02-05:
In my case, renaming INT_MAX to MY_INT_MAX solved issue. Suppose, INT_MAX already defined somewhere. To make clinfo show something, build VC4CL with `cmake -DBUILD_ICD=ON ..`

closed 2024-07-02 by jjsshh3116