doe300/VC4C #167: Is it possible to run the VC4C compiler on x86_64?

wuqs-net, 2024-11-19:
Thanks for your awesome work! I'm trying to compile a large piece of software which will take very long on the Raspberry Pi. So, I want to run the compiler on x86_64 and then copy the binary files to the Raspberry Pi. Is this possible?

doe300, 2024-11-19:
Yes, I did develop the projects on an x64 machine, so the compiler does run there

wuqs-net, 2024-11-19:
Thank you for your reply! To do this, should I enable the `CROSS_COMPILE` flag? It looks like this flag is used to compile VC4C on x64, but I want to run VC4C on x64 to compile OpenCL program for the Raspberry Pi. I might be misunderstanding something, so please correct me.

doe300, 2024-11-19:
If you want to compile on x64 for x64 you don't need the `CROSS_COMPILE` flag. It is only required if you want to compile the VC4C compiler itself on x64 for arm32.
For compiling the OpenCL code with VC4C it makes no difference on which platform the VC4C compiler is run.