doe300/VC4C #168: Build fails on RPi 3B+ Debian Bookworm

kijekijikokwe, 2025-02-21:
Cmake runs fine, but make fails at 91% completion. Would love to get this running as it's supposed to work on RPi 3B+ right?
(Pretty sure I used the Raspbian Lite image to build my OS but my memory is not 100%).

admin@raspberrypi:~/OpenCL_GPU/VC4C/build $ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

admin@raspberrypi:~/OpenCL_GPU/VC4C/build $ uname -a
Linux raspberrypi 6.6.51+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.51-1+rpt3 (2024-10-08) aarch64 GNU/Linux

admin@raspberrypi:~/OpenCL_GPU/VC4C/build $ clang --version
Debian clang version 14.0.6
Target: aarch64-unknown-linux-gnu
Thread model: posix
InstalledDir: /usr/bin

admin@raspberrypi:~/OpenCL_GPU/VC4C/build $ gcc --version
gcc (Debian 12.2.0-14) 12.2.0

admin@raspberrypi:~/OpenCL_GPU/VC4C/build $ apt list --installed | grep llvm
libllvm14/stable,now 1:14.0.6-12 arm64 [installed,automatic]
libllvm15/stable,now 1:15.0.6-4+b1 arm64 [installed,automatic]
llvm-14-dev/stable,now 1:14.0.6-12 arm64 [installed,automatic]
llvm-14-linker-tools/stable,now 1:14.0.6-12 arm64 [installed,automatic]
llvm-14-runtime/stable,now 1:14.0.6-12 arm64 [installed,automatic]
llvm-14-tools/stable,now 1:14.0.6-12 arm64 [installed,automatic]
llvm-14/stable,now 1:14.0.6-12 arm64 [installed,automatic]

admin@raspberrypi:~/OpenCL_GPU/VC4C/build $ make
[  1%] Performing update step for 'cpplog-project'
From https://github.com/doe300/cpplog
 * branch            HEAD       -> FETCH_HEAD
Already up to date.
[  1%] No patch step for 'cpplog-project'
[  1%] Performing configure step for 'cpplog-project'
-- Configuring done
-- Generating done
-- Build files have been written to: /home/admin/OpenCL_GPU/VC4C/build/cpplog/src/cpplog-project-build
[  2%] Performing build step for 'cpplog-project'
[ 50%] Built target cpplog
[100%] Built target cpplog-static
[  3%] Built target cpplog-project-build
[ 48%] Built target VC4CC
[ 48%] Built target GetGitCommit
[ 48%] Building CXX object src/CMakeFiles/VC4C.dir/main.cpp.o
[ 48%] Linking CXX executable vc4c
[ 48%] Built target VC4C
[ 50%] Built target qpu_emulator
[ 88%] Built target TestData
[ 89%] Performing update step for 'cpptest-lite-project'
From https://github.com/doe300/cpptest-lite
 * branch            HEAD       -> FETCH_HEAD
Already up to date.
[ 89%] No patch step for 'cpptest-lite-project'
[ 89%] Performing configure step for 'cpptest-lite-project'
-- Configuring done
-- Generating done
-- Build files have been written to: /home/admin/OpenCL_GPU/VC4C/build/cpptest-lite/src/cpptest-lite-project-build
[ 90%] Performing build step for 'cpptest-lite-project'
[100%] Built target cpptest-lite
[ 91%] Built target cpptest-lite-project-build
[ 91%] Building CXX object test/CMakeFiles/TestVC4C.dir/TestInstructions.cpp.o
<command-line>: warning: "_GNU_SOURCE" redefined
[... 60 lines cut ...]
/home/admin/OpenCL_GPU/VC4C/src/intermediate/../asm/OpCodes.h:861:28: note: candidate: ‘vc4c::PrecalculatedValue vc4c::OpCode::operator()(vc4c::Literal, vc4c::Literal, vc4c::DataType) const’
  861 |         PrecalculatedValue operator()(Literal firstOperand, Literal secondOperand, DataType resultType) const;
      |                            ^~~~~~~~
/home/admin/OpenCL_GPU/VC4C/src/intermediate/../asm/OpCodes.h:861:28: note:   candidate expects 3 arguments, 2 provided
/home/admin/OpenCL_GPU/VC4C/src/intermediate/../asm/OpCodes.h:862:29: note: candidate: ‘vc4c::PrecalculatedVector vc4c::OpCode::operator()(const vc4c::SIMDVector&, const vc4c::SIMDVector&) const’
  862 |         PrecalculatedVector operator()(const SIMDVector& firstOperand, const SIMDVector& secondOperand) const;
      |                             ^~~~~~~~
/home/admin/OpenCL_GPU/VC4C/src/intermediate/../asm/OpCodes.h:862:58: note:   no known conversion for argument 1 from ‘int’ to ‘const vc4c::SIMDVector&’
  862 |         PrecalculatedVector operator()(const SIMDVector& firstOperand, const SIMDVector& secondOperand) const;
      |                                        ~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~
/home/admin/OpenCL_GPU/VC4C/src/intermediate/../asm/OpCodes.h:863:30: note: candidate: ‘vc4c::analysis::ValueRange vc4c::OpCode::operator()(const vc4c::analysis::ValueRange&, const vc4c::analysis::ValueRange&) const’
  863 |         analysis::ValueRange operator()(
      |                              ^~~~~~~~
/home/admin/OpenCL_GPU/VC4C/src/intermediate/../asm/OpCodes.h:864:41: note:   no known conversion for argument 1 from ‘int’ to ‘const vc4c::analysis::ValueRange&’
  864 |             const analysis::ValueRange& firstRange, const analysis::ValueRange& secondRange) const;
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
/home/admin/OpenCL_GPU/VC4C/src/intermediate/../asm/OpCodes.h:865:27: note: candidate: ‘vc4c::OperationBitMasks vc4c::OpCode::operator()(vc4c::BitMask, vc4c::BitMask) const’
  865 |         OperationBitMasks operator()(BitMask firstMask, BitMask secondMask) const;
      |                           ^~~~~~~~
/home/admin/OpenCL_GPU/VC4C/src/intermediate/../asm/OpCodes.h:865:46: note:   no known conversion for argument 1 from ‘int’ to ‘vc4c::BitMask’
  865 |         OperationBitMasks operator()(BitMask firstMask, BitMask secondMask) const;
      |                                      ~~~~~~~~^~~~~~~~~
make[2]: *** [test/CMakeFiles/TestVC4C.dir/build.make:188: test/CMakeFiles/TestVC4C.dir/TestInstructions.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:525: test/CMakeFiles/TestVC4C.dir/all] Error 2
make: *** [Makefile:166: all] Error 2

doe300, 2025-02-21:
It looks like my (badly named) value `INT_MAX` in `/home/admin/OpenCL_GPU/VC4C/test/TestInstructions.cpp:827` conflicts with the `INT_MAX` C standard library macro, which gets pulled in in your compiler. So an immediate fix could be to rename this variable.

Another thing, you seem to build on/for 64-bit ARM. The VC4CL was only ever built for 32-bit system, I highly doubt the hardware access will work, since the system interface most likely is very incompatible...

kijekijikokwe, 2025-02-22:
If you fix the variable, I'll wipe my RPi 3B+, reinstall 32-bit Raspbian (or would Ubuntu be preferable?) and try again. Okay?
Thanks,
Charlotte

doe300, 2025-02-23:
May I ask for what do you want to use this project?
I ask because I abandoned the whole VC4CL project some time ago and I have no intention of taking it up again. Also, I doubt it will be actually useful in any production capacity...

subspecs, 2025-09-27:
@doe300 How about wanting to put a car radiator on it and see how fast this thing can really go.
Would love to get it working in 64bits. Tho, honestly, if you'd provide compiled opencl .so libraries of the opencl 32bit driver along with the opencl headers for this thing then that be great. Because then one could just compile with 32bits in gcc and use it per normal. But I guess that's too much to ask I think